oresat/low-gain-radio #11: Add crystal to the Kw0x  M0 core so it's not dependent on the SX1233 radio clock out.

andrewgreenberg, 2016-06-06:
... the interdependency is a pain, especially for debugging.

wrh2, 2016-10-04:
This is not a good idea because for flight we will need a swept quartz crystal which, as far as I know, are very expensive. Googling around to double check myself, I couldn't find a price which is normally a bad sign. If I am wrong and we can get one cheap, then yes adding another xtal will save us some pain in the firmware. Otherwise, we just need to deal with it appropriately in the firmware.

andrewgreenberg, 2016-10-09:
So it was easy to add another xtal to the layout, so I did it. We don't have to use it, but it's there if we want it. Closing this issue!

closed 2016-10-09 by andrewgreenberg

wrh2, 2016-10-09:
I feel like this is going to encourage to not develop our firmware properly because losing the external clock signal is a possibility during flight regardless of it coming from the XTAL on the transceiver block or not.